sosandroid/FRAM_MB85RC_I2C #3: Testing as much memory devices as possible

sosandroid, 2016-11-18:
I need some help to test the memory devices as I do not have all of them.

The strategy:

- Verifying the device has / has not the Device IDs feature
- Testing the proper adressing scheme by using Manual Mode example
- If device has device IDs feature, check if this is recognized by the lib to set it up automatically.
- Additionnal testing with all the methods of the lib.

Thank you for your help.

Palatis, 2017-01-22:
works for MB85RC16V :-)
thx for the hard work :-D

mrstew, 2017-03-05:
I just tested the library on the FM24C64B - works great!

No device ID, but initialized with the correct chip density (64) and all was good.

sosandroid, 2017-03-09:
Thanks for those feedbacks

JRDavisUF, 2017-03-27:
W10 + CCSv7+ Energia18 + MSP432 LaunchPad + MB85RC256V breakout from Adafruit +v1.2 libray works after a few minor mods:

1) Due to a compiler error, had to modify the line taken from an example:
                FRAM_MB85RC_I2C mymemory();     to
                FRAM_MB85RC_I2C mymemory;

2)  Need to edit this line in getDeviceIDs (this issue is talked about elsewhere online)
          result = Wire.endTransmission(false);    and replace with
          result = ERROR_0;

3)  When doing timing tests, needed to add some Serial.flush() lines before/after timing sections to get stability with results.  I2C with Serial terminal-type output has caused me some issues elsewhere...I think this is related.

Timing Info:
100 bytes array write: 1952 us
100 bytes array read:  2103 us

sosandroid, 2017-03-27:
@JRDavisUF Thx for those feedbacks

MisterScience5, 2017-11-25:
Tested the Cypress FM24CL64B on a Particle Photon, works great in manual mode, initialized as:
FRAM_MB85RC_I2C memory(0x50, false, 13, 64);

hyperion11, 2017-12-20:
test FM24C04B on Arduino stm32. Tests work fine.

Mythricia, 2018-04-30:
Tested FUJITSU MB85RC04V, on Arduino Uno. Seems to work perfectly! Thanks for this great library.

sosandroid, 2018-05-01:
Thank you for sharing

suremicro, 2018-06-16:
Tested Cypress FM24CL04B (4k) on ESP8266 12e (NodeMCU) and works fine.

(When reading all memory 1 byte at a time the ESP soft reset after around 220 reads which was probably due to heap/stack issues. Reading 8 byte arrays works fine.)

ekuiter, 2020-03-01:
I tested the Cypress FM24CL16B with an ESP32. Works like a charm.

sosandroid, 2020-03-31:
Thank you for your feedback

zebra1993, 2021-01-22:
Im currently trying to get a FM24C04B running on my Teensy4.0 Surprisingly, on the "FRAM_I2C_manual_mode.ino" example the FRAM gets detected and is write/readable. when i use it in my normal code i do get the issue of the code not beeign able to talk to it. Could this be from the default I2C interface the wirelibary uses?
https://pastebin.com/UsEqXtxL The output from VSCode if it helps

sosandroid, 2021-01-26:
Good news
Thanks for the feedback.
The device was reported as supported

hugovantriet, 2021-05-25:
Hello,

Thank you really much for making this library, you can't imagine how happy I was when I found this page!
I can confirm that the MB85RC64TA passes the first reading/writing test program!
I'm going to play with it for a bit, and when I find something interesting I will report it back to you.

OptifySudarshanPatil, 2022-12-30:
FM24V05 generates followings error in example code FRAM_I2C
_readIDs

Starting...
....... .......
FRAM Device IDs
Manufacturer 0xFFF
ProductID 0xFFF
Density code 0xF
...... ...... ......