sosandroid/FRAM_MB85RC_I2C #4: Any plan to support this chip in SPI version ?

christophepersoz, 2016-12-22:
Hello,

Do you plan to expand your library to have a SPI version of those chips supported ?
Thanks.

sosandroid, 2017-01-05:
Hi, at this time no plan to support SPI.
However, I can support you if you try to do it

christophepersoz, 2017-01-05:
Hi,

Thanks a lot for the answer. I'm currently making a version for SPI chips, with less functionality and forked from the I2C version - [here](https://github.com/christophepersoz/FRAM_MB85RS_SPI). At this time it's ok, but I will tell you if I need help ;)
Thanks a lot.

christophepersoz, 2017-01-09:
Hello sosandroid,

I'm currently working on the SPI version of those chip. The lib works well, but I would like to simplify it and change the Overloading syntax into a Template. By doing that I, and maybe you too, have only have one write/read function whatever the value you want to write byte, half word, long. The idea is to have a something like this :

```c++
// Inside the .h

template <typename T>
void _readT(T *value) { *value = SPI.tranfer(0); // maybe that can works too with Wire.transmission }

class FRAM_MB85RS_SPI
{
  public:
    boolean read (uint32_t address, T *value);
    // ....

  private:
   //...
}

// Inside the .cpp
boolean FRAM_MB85RS_SPI::read(uint32_t address, T *value)
{
    if (address >= _maxaddress)
        return false;
    
#ifdef SER_OUT
    Serial.print("Read address : ");
    Serial.println(address, BIN);
#endif

    _csASSERT();
        SPI.transfer(FRAM_READ);
        _setMemAddr(& address);
        _readT<uint8_t>(*value);
    _csRELEASE();
    
    return true;
}
```

The point is that I'm a bit lost and I don't have a clue how can I insert a template inside the class.
If you are interested and if you know how to do, maybe you could help me to achieve that ?

The example above doesn't compile and create error on build.

Thanks in advance for your answer

sosandroid, 2017-01-09:
Hi,
I never worked with templates. 
I quick search lead me to issues using Templates with the Arduino Builder such as this one : https://github.com/arduino/arduino-builder/pull/182#issuecomment-264142984

sosandroid, 2017-01-09:
Juste pour info, j'avais fait une autre Lib qui supportait I2C et SPI en même temps. Cela peut servir d'inspiration : https://github.com/sosandroid/AnalogDevice_AD5161

christophepersoz, 2017-01-09:
This SPI lib I made works well, I would just like to minimize the size of the lib in memory by creating something more versatile without adding too much lines of code. Template seems the solution, but hard to figure it out.

christophepersoz, 2017-01-17:
Hello, here the lib I created for the SPI version and based on your work for the I2C FRAM.
Even if I changed a lot of stuff, removed the ERROR management which, wasn't needed for my own use, functions, etc the architecture remain the same.

I hope that it could be useful for the community.

[On GitHub](https://github.com/christophepersoz/FRAM_MB85RS_SPI )
[On GitHub.io page](https://christophepersoz.github.io/FRAM_MB85RS_SPI/)

Thanks for your work!

closed 2017-01-17 by christophepersoz

sosandroid, 2017-01-17:
Just added a crosslink on top of the Readme file to point out this SPI version.
Thank for the sharing