sosandroid/FRAM_MB85RC_I2C #9: Struct limit on the ESP8266?

adbensi, 2017-01-22:
Hello!
Nice job, congradulations!
I have one issue on the ESP8266, I receive only 32 chars.. 

----
Debug
----
ProductID 0x510
Density code 0x5
Density 256K
Device identfied automatically
...... ...... ......
...... ...... ......
...... ...... ......
 mydata :
7
7
8
9
9
10
10
10
11
12
13
13
14
15
15
16
16
16
17
18
18
18
19
19
20
20
20
21
22
22
Init Done - array loaded
...... ...... ......
Calculated address 0x50
Write Done - array loaded in FRAM chip
...... ...... ......
Calculated address 0x50
Read Done - array loaded with read data
...... ...... ......
7
7
8
9
9
10
10
0
255
255
255
255
255
255
255
255
255
[... 605 lines cut ...]
     Serial.println(readdata.datastruct.S16H,DEC);
     Serial.println(readdata.datastruct.S17H,DEC);
     Serial.println(readdata.datastruct.S18H,DEC);
     Serial.println(readdata.datastruct.S19H,DEC);
     Serial.println(readdata.datastruct.S20H,DEC);
     Serial.println(readdata.datastruct.S21H,DEC);
     Serial.println(readdata.datastruct.S22H,DEC);
     Serial.println(readdata.datastruct.S23H,DEC);
     Serial.println(readdata.datastruct.S24H,DEC);
     Serial.println(readdata.datastruct.S25H,DEC);
     Serial.println(readdata.datastruct.S26H,DEC);
     Serial.println(readdata.datastruct.S27H,DEC);
     Serial.println(readdata.datastruct.S28H,DEC);
     Serial.println(readdata.datastruct.S29H,DEC);
     Serial.println(readdata.datastruct.S30H,DEC);    
 
	Serial.println("Read Write test done - check data if successfull");
	Serial.println("...... ...... ......");	

	
}

void loop() {
	// nothing to do
}

Palatis, 2017-01-22:
TWI buffer is limitted to 32 bytes. same on AVR.

I managed to solve the problem by breaking the reads / writes:
```
#include <utility/twi.h>

void _readBytes(FRAM_MB85RC_I2C & fram, uint16_t addr, uint8_t length, uint8_t * buffer) {
	while (length != 0) {
		uint8_t bytes_to_read = min(length, TWI_BUFFER_LENGTH);
		fram.readArray(addr, bytes_to_read, buffer);
		addr += bytes_to_read;
		buffer += bytes_to_read;
		length -= bytes_to_read;
	}
}

void _writeBytes(FRAM_MB85RC_I2C & fram, uint16_t addr, uint8_t length, uint8_t *buffer) {
	while (length != 0) {
		uint8_t bytes_to_write = min(length, TWI_BUFFER_LENGTH);
		fram.writeArray(addr, bytes_to_write, buffer);
		addr += bytes_to_write;
		buffer += bytes_to_write;
		length -= bytes_to_write;
	}
}
```

adbensi, 2017-01-23:
Thanks a lot! I test 1Mbit memory and 256Kbit memory,
1Mbit memory works when I change the Product ID
  /* 0x758 = 1M and 0x510 = 256K*/
  if ((prodID != 0x758)&&(prodID != 0x510))
  {
    Serial.print("Unexpected Product ID: 0x");
    Serial.println(prodID, HEX);
    return false;
  }
May be it was usefull to you,
Thanks a lot again,
Best Regards!
Alexandre

MisterScience5, 2017-11-26:
Hey, I am also having these issues, and due to my lack of experience with C I'm having some issues getting Palatis' script that iterates through chunks of it working...

Any tips would be greatly appreciated:)

JoeLoginIsAlreadyTaken, 2018-01-14:
I got a FM24CL16 (connected to a ESP8266), from Aliexpress, to work using Palati's approach with a fixed chunk size of 16 bytes.
"TWI_BUFFER_LENGTH" seems to be not declared on ESP8266.
Using a size of 32 bytes results in one lost byte.

Here is my modified "store_anything"example:

```
/**************************************************************************/
/*!
    @file     FRAM_I2C_store_anything.ino
    @author   SOSAndroid (E. Ha.)
    @license  BSD (see license.txt)

    Example sketch that stores and retreive any kind of data.

    @section  HISTORY

    v1.0 - First release
	v1.0.1 - fix constructor call error
*/
/**************************************************************************/

#include <Wire.h>
#include <math.h>

#include <FRAM_MB85RC_I2C.h>

//define a struct of various data types
typedef struct MYDATA_t {
	bool data_0;
	float data_1; 
	long data_2; 
	int data_3;
	byte data_4;
  char data_5[64];
};

//define a struct joining MYDATA_t to an array of bytes to be stored
typedef union MYDATA4I2C_t {
 MYDATA_t datastruct;
 uint8_t I2CPacket[sizeof(MYDATA_t)];
};

MYDATA4I2C_t mydata; //data to be written in memory
MYDATA4I2C_t readdata; //data read from memory

//random address to write from
uint16_t writeaddress = 0x01;

//Creating object for FRAM chip
FRAM_MB85RC_I2C mymemory(MB85RC_ADDRESS_A000, false,D3,16);

byte _readBytes(FRAM_MB85RC_I2C & fram, uint16_t addr, uint8_t length, uint8_t * buffer) {
  byte result=0;
  while (length != 0) {
    uint8_t bytes_to_read = min(length, 16);

    result = fram.readArray(addr, bytes_to_read, buffer);
    if(result>0) return result;
    addr += bytes_to_read;
    buffer += bytes_to_read;
    length -= bytes_to_read;
 
  }
  return result;
}

byte _writeBytes(FRAM_MB85RC_I2C & fram, uint16_t addr, uint8_t length, uint8_t *buffer) {
  byte result=0;
  while (length != 0) {
    uint8_t bytes_to_write = min(length, 16);
    result = fram.writeArray(addr, bytes_to_write, buffer);
    if(result>0) return result;
    addr += bytes_to_write;
    buffer += bytes_to_write;
    length -= bytes_to_write;
  }
   return result;
}

void setup() {

	Serial.begin(115200);
	while (!Serial) ; //wait until Serial ready
	Wire.begin();
	
	byte arraySize = sizeof(MYDATA_t);
	
    Serial.println("Starting...");
		
	mymemory.begin();
		
	
//---------init data - load array
	mydata.datastruct.data_0 = true;
	Serial.print("Data_0: ");
	if (mydata.datastruct.data_0) Serial.println("true");
	if (!mydata.datastruct.data_0) Serial.println("false");
	mydata.datastruct.data_1 = 1.3575;
	Serial.print("Data_1: ");
	Serial.println(mydata.datastruct.data_1, DEC);
	mydata.datastruct.data_2 = 314159L;
	Serial.print("Data_2: ");
	Serial.println(mydata.datastruct.data_2, DEC);
	mydata.datastruct.data_3 = 142;
	Serial.print("Data_3: ");
	Serial.println(mydata.datastruct.data_3, DEC);	
	mydata.datastruct.data_4 = 0x50;
	Serial.print("Data_4: 0x");
	Serial.println(mydata.datastruct.data_4, HEX);
  String str = "ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz";
  str.toCharArray(mydata.datastruct.data_5, 64);
  Serial.print("Data_5: ");
  Serial.println(mydata.datastruct.data_5);
	
	Serial.println("...... ...... ......");
	Serial.println("Init Done - array loaded");
	Serial.println("...... ...... ......");

//----------write to FRAM chip
	byte result = _writeBytes(mymemory,writeaddress, arraySize, mydata.I2CPacket);
  Serial.print("Array Size=");
  Serial.println(arraySize);

    if (result == 0) Serial.println("Write Done - array loaded in FRAM chip");
    if (result != 0) Serial.println("Write failed");
	Serial.println("...... ...... ......");
	
	
//---------read data from memory chip
	result = _readBytes(mymemory,writeaddress, arraySize, readdata.I2CPacket);
    if (result == 0) Serial.println("Read Done - array loaded with read data");
    if (result != 0) Serial.println("Read failed");
	Serial.println("...... ...... ......");

  
//---------Send data to serial
	Serial.print("Data_0: ");
	if (readdata.datastruct.data_0) Serial.println("true");
	if (!readdata.datastruct.data_0) Serial.println("false");
	Serial.print("Data_1: ");
	Serial.println(readdata.datastruct.data_1, DEC);
	Serial.print("Data_2: ");
	Serial.println(readdata.datastruct.data_2, DEC);
	Serial.print("Data_3: ");
	Serial.println(readdata.datastruct.data_3, DEC);	
	Serial.print("Data_4: 0x");
	Serial.println(readdata.datastruct.data_4, HEX);
  Serial.print("Data_5: ");
  Serial.println(readdata.datastruct.data_5);
	Serial.println("...... ...... ......");
	Serial.println("Read Write test done - check data if successfull");
	Serial.println("...... ...... ......");	
	
}

void loop() {
	// nothing to do
}
```

closed 2018-02-26 by sosandroid

Palatis, 2018-03-06:
> "TWI_BUFFER_LENGTH" seems to be not declared on ESP8266.

https://github.com/esp8266/Arduino/blob/master/libraries/Wire/Wire.h#L32