sosandroid/FRAM_MB85RC_I2C #14: FM24V10 not recognized

pastaclub, 2018-02-22:
I tested the library with a Cypress FM24V10 and the device is not recognized. The debug output reports "Chip not found".

I made sure the chip is working, by talking to it through the Wire library (not using this library). I can successfully read and write bytes. The device ID returned by the chip is 0x004400.

sosandroid, 2018-02-22:
Hi,

Thanks for this feedback. The devices had not been tested so far. Is there a way to test with the example FRAM_I2C_readIDs.ino and post the results ? This would help to enhance the lib.

thx

sosandroid, 2018-02-26:
After doucble checking the datasheet, the lib uses the proper settings to recognize the chip : manufacturerID : 0x004 & Density Code 0x04.

Your help testing the chip with the the readIDs example would is appreciated. How does the lib behave using manual mode ?

regards

pastaclub, 2018-02-27:
I ended up writing my own library since all I needed was reading and writing bytes.

From my own experiments with the chip I know for sure that when requesting the device ID, the three bytes returned are 0x00, 0x44 and 0x00. When trying to make the library work, I had a look at the code and it looked correct (the constants related to manufacturer and density are correct and the bit shifting should also retrieve the values from the right location). Not sure why it didn't work. I only tried the FRAM_I2C_connect example.

I went to the seaside for a few days so I cannot test now, but I'll have another look when I get back home.

pastaclub, 2018-03-04:
I ran `FRAM_I2C_readIDs.ino`. It's output is:
`Starting...
....... .......
FRAM Device IDs
Manufacturer 0x4
ProductID 0x400
Density code 0x4
...... ...... ......`

pastaclub, 2018-03-04:
I ran the connect example again and now magically it works:
```
Starting...
FRAM_MB85RC_I2C object created
I2C device address 0x50
WP pin number 13
Write protect management: false
Memory Chip initialized
FRAM Device IDs
Manufacturer 0x4
ProductID 0x400
Density code 0x4
Density 1024K
Device identfied automatically
...... ...... ......
...... ...... ......
```

I have no clue what prevented it from working before, since it's connected in the same way. It is particularly strange that the library at first reported "Chip not found", but manual calls to `Wire.write` allowed me to correctly retrieve the ID and read and write. A vague suspect is that I was not using pullups on the I2C lines and that sometimes leads to unpredictable behavior.

I guess we can close this as a false alarm and add FM24V10 to the list of devices detected properly.

closed 2018-03-04 by pastaclub

sosandroid, 2018-03-04:
Thanks for the feedback. I'll marke the chip as tested OK.