sosandroid/FRAM_MB85RC_I2C #15: Test FM24C04B-G

bobybobybob, 2018-03-02:
**FM24C04B-G**

Starting...
FRAM_MB85RC_I2C object created
I2C device address 0x50
WP pin number 13
Write protect management: false
Memory Chip initialized
FRAM Device IDs
Manufacturer 0xF00
ProductID 0xF00
Density code 0xF00
Density 4K
Device properties set
...... ...... ......
...... ...... ......
1st test
Writing at location 0x25
Writing...
Calculated address 0x0
Reading...
Calculated address 0x0
Written value 0xBE
Read value 0xBE
Write Byte test : OK
.... ....
2nd test
Writing at location 0x1B0
Writing...
Calculated address 0x0
Reading...
Calculated address 0x0
Written value 0xBEEF
Read value 0xBEEF
Write word test : OK
.... ....
Manufacturer 0xF00
DensityCode 0xF00
Density 4

sosandroid, 2018-03-04:
Hi,
I suppose this means successfull test :)

bobybobybob, 2018-03-04:
yes you can write it is ok on startpage.

sosandroid, 2018-03-04:
Thanks for your help

closed 2018-03-04 by sosandroid