sosandroid/FRAM_MB85RC_I2C #21: Is this quicker for writing arrays?

benlucy, 2019-10-28:
Hi,

As I understand it, the adafruit library writes arrays inefficiently because it takes 4 bytes to write a single byte: https://arduino.stackexchange.com/questions/44082/i2c-fram-read-write-is-very-slow-taking-over-500-microseconds-per-read-almost

Is this library faster? Could you estimate how long it would take to write an array of 240 bytes?

Thanks

sosandroid, 2019-10-28:
To be honest, I don't know.

The test depends on your environment : board, device, bus speed underlying TWI library, and so on....

Despite this lib has been inspired by the one from adafruit, this is in fact a really different one.

The only way to know is to test it on your side

sosandroid, 2019-10-31:
From what I read on the link you provided, writing array may be quicker.
This lib set the address pointer and send all arrays data. It has been designed to consider everything as an array.

But the best way is to test on your side

benlucy, 2019-11-09:
Thanks for that. It turns out that FRAM is not what I need (there is GRAM in a screen I can use), so I won't be testing or updating sorry.

sosandroid, 2019-11-10:
Welcome

closed 2019-11-10 by sosandroid