sosandroid/FRAM_MB85RC_I2C #23: Cypress FM24C04B multiple issues

Phil-GS, 2020-01-23:
This FRAM is being used with this library
Issue 1 in Simple Read Write sketch from examples
Compile error "FRAM_I2C_simple_write_read:52:21: error: request for member 'writeByte' in 'mymemory', which is of non-class type 'FRAM_MB85RC_I2C()'"
Issue 2 None of the other sketches return values or report failed to write and read
Nick Gammons Address finder sketch does not find devices. Some of the FRAM sketches appear to find device on 0x50.
As reported in a previous issue, the chip does work with the Arduino IDE EEPROM examples and successfully clears, writes and reads, so as far as I know, chip not faulty and wiring correct.
Any thoughts please?

Phil-GS, 2020-01-23:
Just looked at bobbybobbybob issue
Mine reports exactly the same output, as as author states, I am also using the examples as is.
I might try a 64kb version of the chip

Phil-GS, 2020-01-23:
Didn't mean to use comment and close

sosandroid, 2020-01-23:
Hi,
Can you please provide more detailed data about the compile error on a specific issue ?

sosandroid, 2020-01-23:
The Cypress module chip does not have the IDs feature according to datasheet

sosandroid, 2020-01-23:
Do you have a link to the quoted sketch ?

I will have a look when back from my business trip

sosandroid, 2020-01-23:
If same issue as #18, please try manual mode

Phil-GS, 2020-01-23:
Hello

Thanks for replying.

I will deal with these problems in the order they arrived here.

Basically, I bought a Cypress FM24C04B and FM24C64B chip as you had them listed as tested.

Soldered onto a SOIC-8 adaptor, wired up to a Nano 328P.

I had the library installed and just worked through the examples one by one.

 

This is the compile error listing for the simple write/read sketch

 

C:\Users\Phil\Documents\Arduino\libraries\FRAM_MB85RC_I2C\examples\FRAM_I2C_simple_write_read\FRAM_I2C_simple_write_read.ino: In function 'void setup()':

FRAM_I2C_simple_write_read:44:11: error: request for member 'begin' in 'mymemory', which is of non-class type 'FRAM_MB85RC_I2C()'

  mymemory.begin();

           ^~~~~

FRAM_I2C_simple_write_read:52:21: error: request for member 'writeByte' in 'mymemory', which is of non-class type 'FRAM_MB85RC_I2C()'

  resultw = mymemory.writeByte(writeaddress, writevalue);

                     ^~~~~~~~~

FRAM_I2C_simple_write_read:54:21: error: request for member 'readByte' in 'mymemory', which is of non-class type 'FRAM_MB85RC_I2C()'

  resultr = mymemory.readByte(writeaddress, &readvalue);

                     ^~~~~~~~

FRAM_I2C_simple_write_read:73:21: error: request for member 'getOneDeviceID' in 'mymemory', which is of non-class type 'FRAM_MB85RC_I2C()'

  resultr = mymemory.getOneDeviceID(1, &id);

                     ^~~~~~~~~~~~~~

FRAM_I2C_simple_write_read:77:21: error: request for member 'getOneDeviceID' in 'mymemory', which is of non-class type 'FRAM_MB85RC_I2C()'

  resultr = mymemory.getOneDeviceID(3, &id);

                     ^~~~~~~~~~~~~~

FRAM_I2C_simple_write_read:81:21: error: request for member 'getOneDeviceID' in 'mymemory', which is of non-class type 'FRAM_MB85RC_I2C()'

  resultr = mymemory.getOneDeviceID(4, &id);

                     ^~~~~~~~~~~~~~

 

Multiple libraries were found for "Wire.h"

Used: C:\Users\Phil\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.2\libraries\Wire

Multiple libraries were found for "FRAM_MB85RC_I2C.h"

Used: C:\Users\Phil\Documents\Arduino\libraries\FRAM_MB85RC_I2C

exit status 1

request for member 'begin' in 'mymemory', which is of non-class type 'FRAM_MB85RC_I2C()'

 

That’s as far as that one went

Phil (S)

Phil-GS, 2020-01-23:
Hello

Next issue

I have the Cypress datasheet for the 4-kb chip I am using.

It has 7-bits for the address, from MSB, the first four are the device identifier, the next two are A1 and A2 (both of mine grounded, so 0,0), the 7th bit is A0 which the 4-kb chip does not have (pin 1 N/C), and the 8th bit LSB is R/W.

To quote the datasheet

Slave Device Address

The first byte that the FM24C04B expects after a START

condition is the slave address. As shown in Figure 6, the slave

address contains the device type or slave ID, the device select

address bits, a page select bit, and a bit that specifies if the

transaction is a read or a write.

Bits 7-4 are the device type (slave ID) and should be set to 1010b

for the FM24C04B. These bits allow other function types to

reside on the I2C bus within an identical address range. Bits 3-2

are the device select address bits. They must match the corresponding

value on the external address pins to select the device.

Up to four FM24C04B devices can reside on the same I2C bus

by assigning a different address to each. Bit 1 is the page select

bit. It specifies the 256-byte block of memory that is targeted for

the current operation. Bit 0 is the read/write bit (R/W). R/W = ‘1’

indicates a read operation and R/W = ‘0’ indicates a write

operation..

It reads to me, I’m no expert, that it needs the device identifier. The 7-bit address is 1010000 which translates to HEX 0x50

Nick Gammons I2C address finder sketch reported no devices found.

Hope that helps

Phil (S)

Phil-GS, 2020-01-23:
Hello again

Not sure which was the quotes sketch, but this is the output from the manual mode example

 

Starting...

FRAM_MB85RC_I2C object created

I2C device address 0x50

WP pin number 13

Write protect management: false

Memory Chip initialized

FRAM Device IDs

Manufacturer 0xF00

ProductID 0xF00

Density code 0xF00

Density 4K

Device properties set

...... ...... ......

...... ...... ......

1st test

Writing at location 0x25

Writing...

Calculated address 0x50

Reading...

Calculated address 0x50

Written value 0xBE

Read value 0xFF

Write Byte test : NOT OK

.... ....

2nd test

Writing at location 0x1B0

Writing...

Calculated address 0x50

Reading...

Calculated address 0x51

Written value 0xBEEF

Read value 0xFFFF

Write word test : NOT OK

.... ....

Manufacturer 0xF00

DensityCode 0xF00

Density 4

This is exactly the same as the previous correspondent

Phil (S)

Phil-GS, 2020-01-23:
Hello

Please see my last reply.

In a nutshell, none of the examples work with the 4-kb chip.

I have yet to try the 64-kb version.

I’m struggling with this, no expert, but feel it has to be something simple.

Looking at the RTC DS1307 datasheet, another I2C device, it quotes for the device address

 

The slave address byte contains the 7-bit DS1307 address, which is 1101000, followed by the direction bit (R/W), which is 1 for a read. 

 

Which is very similar and is the familiar 0x68 address. The RTC sketches all work OK.

I would like to get the FRAM working as I’m data-logging using the Mega 2560 EEPROM and have already busted the 100,000 writes by a factor of 4 (still works), but the Mega blew up on me so all the data gone, hence the need for external memory with better write endurance.

Just to clarify my setup; FM24C04B pins – pin one N/C; pin 2 A1 ground; pin 3 A2 ground; pin 4 ground; pins 5 and 6 SDA and SCL; pin 7 WP ground or Nano  pin 13 as required; pin 8 5-V

 

What I can’t find is the link to someone who was having problems with the FM24C04, but found it worked with the standard EEPROM library. I can confirm that without any hardware or software changes, I was able to get all the examples I tried working fine – very strange.

Look forward to hearing what you make of it.

Phil (S)

sosandroid, 2020-01-30:
Oups, missed answers due to high level of work last days.
Going through this on smartphone is not ideal. Will try to readout from a laptop asap

sosandroid, 2020-01-30:
Just had a look on datasheet as it is far away from now.
The Cypress has 1010+A2+A1 as address and A0 as page select.
From quick reading, A0 can be 1 for write operation.

Can you check the behaviour when using the address 0b1010001 ?

Phil-GS, 2020-01-30:
Hello

I messed about with this for a while then decided to get an Adafruit breakout board with 256-kb Fujitsu I2C on board to get some restored confidence.

It worked fine with the Adafruit FRAM library.

I wasn’t getting any Cypress I2C devices found with Nick Gammons’ scanner, but after a lot of checking and changing the pullup resistors, it did find not one, but two devices,(with only one on the bus) say 0x54 and 0x55, which might account for a lot of the problems.

Once one address had been selected, one by one the example sketches worked.

The sketch that hadn’t been compiling (might have been simple read write) worked when I borrowed the setup details from Manual mode to fill out the constructor.

I can’t account for two devices found and will live with that.

I noticed in the Adafruit tutorial, they recommend power cycling the FRAM when the address jumpers are changed.

I’m quite happy that these work now, but I will probably use the Fujitsu FRAMs instead of Cypress.

I also got hold of a Cypress FRAM development kit with SPI and I2C devices on it  and will see if it throws up the dual addresses.

I have 4-kb and 64-kb Cypress chips. The 4-kb has A0 as not connected.

Getting hold of Fujitsu in UK is difficult, but I got some from Digikey.

Right now, attention has turned to SD cards, but I will have to implement the FRAM as I reckon I had over 400,000 writes to a Mega 2560 EEPROM which was still working but occasionally getting flaky.

Cheers

sosandroid, 2020-03-31:
Sorry for late reply. Work do not allows me to stick on this Github account.
From my understanding, you find a way to address the Cypress devices but will prfer Fujitsu ones. That might be a good deal.
Do you consider the issue close?

Phil-GS, 2020-04-01:
Hello

I think I got most of the FRAM (Cypress and Fujitsu) working.

Like you, very busy.

Please close.

Thanks

Phil_S

closed 2020-04-02 by sosandroid