sosandroid/FRAM_MB85RC_I2C #24: Build Error on Example

thommaughan, 2020-02-02:
IDE: Arduino 1.8.9
Board: AdaFruit Feather M0
Example: FRAM_I2C_simple_write_read

FRAM_I2C_simple_write_read:44:11: error: request for member 'begin' in 'mymemory', which is of non-class type 'FRAM_MB85RC_I2C()'

  mymemory.begin();

           ^

FRAM_I2C_simple_write_read:52:21: error: request for member 'writeByte' in 'mymemory', which is of non-class type 'FRAM_MB85RC_I2C()'

  resultw = mymemory.writeByte(writeaddress, writevalue);

sosandroid, 2020-02-02:
Looks like the class is not loaded. Did you import it in the libs folder of Arduino IDE ?

thommaughan, 2020-02-02:
I have the library in the directory with all my other libraries.   I did
not use the arduino IDE to import it though.   I have other libraries in
the same folder that are copied in and all seems well.

I have been making progress on adding support for the 128Kx8 Fujitsu
devices by hacking the Adafruit FRAM library (also copied).   I'm a C
programmer, and I'm a bit weak on C++ but I suspect something with the
object instantiation / constructor call

//Creating object for FRAM chip
FRAM_MB85RC_I2C mymemory();

vs the Adafruit way:

Adafruit_FRAM_I2C fram     = Adafruit_FRAM_I2C();

sosandroid, 2020-02-03:
Can you remove the brackets at the object creation ?

````FRAM_MB85RC_I2C mymemory;````

This is the way the example had been modified a few days ago.

sosandroid, 2020-02-03:
Another way is using the constructor providing the chip address and the use write protect

````FRAM_MB85RC_I2C mymemory(address, false);````

Let me know

thommaughan, 2020-02-04:
I'll give it a try when I get a time, thanks for the help

sosandroid, 2020-03-31:
No feedback, I am closing the issue

closed 2020-03-31 by sosandroid