sosandroid/FRAM_MB85RC_I2C #29: How to use Wire1

Jim-bee, 2021-11-07:
I am trying to get the 128Kbit fuji fram working with your library but I am on Wire1 bus.  In your library.cpp file I have tried to change Wire to Wire1 and WP pin from 13 to 21 but I still get Memory Chip NOT FOUND (write/read fails).   I am using your simple_write_read example.  Is there a way to use Wire1 with your library?

Starting...
FRAM_MB85RC_I2C object created
I2C device address 0x50
WP pin number 21
Write protect management: true
Memory Chip NOT FOUND
...... ...... ......
1st test
Writing at location 0x25
Writing...
Reading...
Written value 0xBE
Read value 0x0
Write Byte test : NOT OK
.... ....
Manufacturer 0xFFF
DensityCode 0xF
Density 0

sosandroid, 2021-11-08:
Hello,

More details would help.

Dealing with wire1 sounds like using Due board. There's a lot of issues around this board with I2C not related with this library.

No ways for me to help on that as I am not able to test your setup. I do not have Due nor Fram available at this time.

JensGrabner, 2021-11-27:
Google Search "Wire1 I2C" can help

closed 2023-01-04 by sosandroid