sosandroid/FRAM_MB85RC_I2C #31: FM24C04B problem

rozrabiak, 2023-08-11:
Hi, I have connected FM24C04B.
i2c scanner shows me this addresses:
```
01:07:29.324 -> I2C device found at address 0x50
01:07:29.324 -> I2C device found at address 0x51
```
but I have only one chip connected to Arduino.
Which address should I use?
Library in debug mode shows me:

```
01:10:14.034 > FRAM
01:10:14.055 > FRAM_MB85RC_I2C object created
01:10:14.055 > I2C device address 0x50
01:10:14.055 > WP pin number 13
01:10:14.055 > Write protect management: true
01:10:14.055 > Memory Chip NOT FOUND
01:10:14.055 > ...... ...... ......
```

Thank's for help, reply, etc.

sosandroid, 2023-08-12:
Hi,

Can you check the address ?
Have a look on datasheet to check this out

rozrabiak, 2023-08-12:
Above addresses I got when A1 and A2 pulled-down.
When A1 and A2 are pull-down I have two addresses: 0x7... I don't remember now :-(
In my IC I have two pi for address settings, A1 and A2.

sosandroid, 2023-08-12:
The address is then 0b101000 following by the page bit. This makes a 7bit adresse which would work (8th bit is reserved for R/W operation)

alex-eri, 2024-03-15:
Memory Chip NOT FOUND

same problem

	resultr = mymemory.getOneDeviceID(1, &id);
	Serial.print("Manufacturer 0x");
	Serial.println(id, HEX);
	
	resultr = mymemory.getOneDeviceID(3, &id);
	Serial.print("DensityCode 0x");
	Serial.println(id, HEX);
	
	resultr = mymemory.getOneDeviceID(4, &id);
	Serial.print("Density ");
	Serial.println(id, DEC);
	
returns 

Manufacturer 0xFFF
DensityCode 0xF
Density 0

alex-eri, 2024-03-15:
.....1st test
Writing at location 0x25
Writing...
Calculated address 0x50
Reading...
Calculated address 0x50
[WiFi-event] event: 4
[WiFi-event] event: 7
WiFi connected. IP address: 10.99.99.163
Connecting to MQTT... in loop
[  2632][E][Wire.cpp:392] requestFrom(): i2cRead returned Error 263
Written value 0xBE
Read value 0xFF
Write Byte test : NOT OK
.... ....
2nd test
Writing at location 0x7B0
Writing...
Calculated address 0x50
Reading...
Calculated address 0x57
[  3643][E][Wire.cpp:392] requestFrom(): i2cRead returned Error 263
Written value 0xBEEF
Read value 0xFFFF
Write word test : NOT OK
.... ....

esp32, i have working bmp180 and pcf8574ts on wire

tried wire running at 100000 and 400000

alex-eri, 2024-03-15:
i2c scan detects 0x50 if Vdd = 5V, but not works.

i used 10k pullup resistors near ic